xmos/vocalfusion-rpi-setup #27: Fails at I2S kernel build on I2S on Pi 3B+

ed-xmos, 2021-03-23:
cp ../src/loader.c i2s_master_loader.c
make -C /lib/modules/4.19.97-v7+/build M=/home/pi/vocalfusion-avs-setup/vocalfusion-rpi-setup/loader/i2s_master modules
make[1]: *** /lib/modules/4.19.97-v7+/build: No such file or directory.  Stop.
make: *** [Makefile:9: all] Error 2
Error: I2S kernel module build failed

ed-xmos, 2021-03-23:
The same happens on a 3B:

make CFLAGS_MODULE='-DI2S_MASTER'
# copy the source file to avoid maintaining multiple copies inside the Makefile folders
# We tried to use KBUILD_OUTPUT but the module was not compiling
cp ../src/loader.c i2s_master_loader.c
make -C /lib/modules/4.19.97-v7+/build M=/home/pi/vocalfusion-avs-setup/vocalfusion-rpi-setup/loader/i2s_master modules
make[1]: *** /lib/modules/4.19.97-v7+/build: No such file or directory.  Stop.
make: *** [Makefile:9: all] Error 2
Error: I2S kernel module build failed

ed-xmos, 2021-03-23:
It looks like https://github.com/xmos/vocalfusion-rpi-setup/blob/f08a5888e7b78a100baedde3ba9ff9e4b2842602/setup.sh#L55 is the problem. It downloads for version 5.10 (buster) rather than 4.19

DaveAtkinsonXMOS, 2021-03-30:
Currently we have the vocalfusion-avs-setup repo tied to the 2020-02-13 raspbian buster release, by way of the readme. This because, even if you do get the headers to match, there seems to be a problem with I2S on the latest v5.xx kernel for rapsberry pi os. I didn't realise this repo gets used stand-alone, shall I modify the readme to contain the same information about which setup package to use? (https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/2020-02-13-raspbian-buster.zip)

DaveAtkinsonXMOS, 2021-03-31:
This should be fixed by #30, we now only install the kernle headers once, before anyone has had a chance to call 'apt-get update'.

closed 2021-03-31 by DaveAtkinsonXMOS

DaveAtkinsonXMOS, 2021-03-31:
I've modified the setup.sh script to only install the kernel headers once, before anyone has had a chance to call "apt-get update". You can now run the script multiple times this should be address by #30.