xmos/vocalfusion-rpi-setup #29: DAC setup uses internal PLL based off BCLK but BCLK is bursty - May cause audio distortion at start of playback.

XMOS-JoeG, 2021-03-26:
The Audio DAC setup is such that it is using the BCLK input to generate its own internal master clock using a PLL. This is intended only for when BCLK is free running. In the raspberry pi BCLK is only output when playing audio. This means initially the DAC will take some time to lock to the new BCLK and so audio output may be distorted and/or delayed. Many DACs do not have an internal PLL so this isn't also a great example use case.

Suggest we move to more standard system of using the master clock input directly which is free running. I can provide a new register setup for this.

mbanth, 2022-06-13:
@XMOS-JoeG, is this issue still important to address?

XMOS-JoeG, 2022-06-13:
This is technical debt essentially, it makes no difference to me. It was more a general point of how to do things properly to take note of in future.

lucianomartin, 2022-07-11:
@XMOS-JoeG confirmed that given that the current setup is quite stable, it's probably more risky to modify this now, given we don't have all the information on how we might have used this in that past.
@tomblackie-xmos, can we close this issue or assign to a different deliverable?

lucianomartin, 2022-09-20:
@XMOS-JoeG, if we use the master clock do we need to have different sets of register setups to support audio rates of 16 and 32kHZ? I suppose using BCLK, the audio rate must not be explicitly supplied to the DAC.
.

XMOS-JoeG, 2022-09-20:
This is DAC dependent but in general yes you would need to change the setup of the dac if you changed sample frequency. This applies for whichever method is used to generate the master clock

lucianomartin, 2022-11-10:
@XMOS-JoeG provided a script to solve this issue. See the text from the email below.

```
Hi Luciano,

I’ve attached [some code ]([audio_dac.zip](https://github.com/xmos/vocalfusion-rpi-setup/files/9980178/audio_dac.zip))which sets the DAC up with the following config:
Fs = 48kHz, Master clock taken from MCLK pin @ 12.288MHz, I2S slave mode with data input on DIN pin, 24bit, All volume set to max (0dB).

If you want to change this for Fs = 16kHz then you would need to change the DOSR value to 384 instead of 128. Reg writes to do this are commented out in the code.
Any further setup you require is documented in the DAC datasheet.

Cheers,
Joe
```

mbanth, 2023-09-15:
Now tracked at [Carried Enhancements](https://xmosjira.atlassian.net/wiki/spaces/TNPMS/pages/3870916609/Carried+Enhancements).

closed 2023-09-15 by mbanth