xmos/vocalfusion-rpi-setup #50: setup.sh fails due to missing package

ed-xmos, 2022-10-25:
This happened on a 3B and 4. I found I could work around it by removing audacity from the required packages to allow setup.sh to complete.

`Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  audacity-data libflac++6v5 libportaudio2 libportsmf0 libsuil-0-0 libvamp-hostsdk3v5 libwxbase3.0-0v5 libwxgtk3.0-0v5
Suggested packages:
  ladspa-plugin
The following NEW packages will be installed:
  audacity audacity-data libflac++6v5 libportaudio2 libportsmf0 libsuil-0-0 libvamp-hostsdk3v5 libwxbase3.0-0v5 libwxgtk3.0-0v5
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 34.7 kB/9,230 kB of archives.
After this operation, 36.5 MB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libflac++6v5 armhf 1.3.2-3+deb10u1
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/f/flac/libflac++6v5_1.3.2-3+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Error: installation of package audacity failed after 11 attempts
Please retry installation procedure.`

lucianomartin, 2022-11-29:
To fix this issue we need to remove the instrution:
`DO NOT FOLLOW THE PROMPT TO UPDATE THE SOFTWARE ON THE SYSTEM. Set up the locale, and setup a network connect, but DO NOT update the software on the Raspberry Pi. This will update the kernel, and then the audio sub-system will not work.`
from https://github.com/xmos/vocalfusion-avs-setup/blob/master/README.md.
As far as I know, this can be done when we fix the I2S clocks as described in https://github.com/xmos/sw_xvf3800/issues/134.
If we remove that information, we need to periodically check that the setup works, as we will rely on the latest version of the Rasbpian packages

lucianomartin, 2023-03-28:
Fixed by https://github.com/xmos/vocalfusion-rpi-setup/releases/tag/v5.7.0

closed 2023-03-28 by lucianomartin