xmos/vocalfusion-rpi-setup #52: What is the problem with I2S sub-system on RPi OS with Linux kernel v5?

jfathman, 2022-12-08:
Excerpts from vocalfusion-rpi-setup README.md on 8 Dec 2022:

    We cannot use the latest as updates to linux kernel v5 have broken the I2S sub-system.

    **DO NOT** follow the prompt to update the software on the system. Set up the locale,
    and setup a network connect, but **DO NOT** update the software on the Raspberry Pi.
    This will update the kernel, and then the audio sub-system will not work.

I am running the latest Raspberry Pi OS 64-bit Bullseye 11.5 with Linux kernel version 5.15.76-v8+ on RPi 3 B+ connected to XVF3000 hardware, and audio (play and record) work properly, with good audio quality.

Is it surprising that audio works for my configuration?  What is the expected I2S problem that produced the README.md warning notes?  What should I look for to know if there is a problem lurking that I have not yet observed?

Thanks!

jfathman, 2022-12-15:
Raspberry Pi OS 64-bit Bullseye 11.5 with Linux kernel version 5.15.76-v8+ connected to XVF3000 hardware
also works on RPi 4 B (with the XMOS I2S module built with RPI_4B defined).

fquirin, 2023-03-10:
The XVF3510-INT build on Raspberry Pi OS 64-bit Bullseye with kernel 5.15.84-v8+ seems to work as well 🤷.

PR #63 (2023-03-22): Update raspbian image in README file
Closes #52
Closes https://github.com/xmos/sw_xvf3800/issues/689
PR #65 (2023-03-23): Release/v5.7.0
  * Update Raspbian image to Bullseye (2023-02-21-raspios-bullseye-armhf)
  
  Closes #52
closed 2023-03-23 by lucianomartin