xmos/vocalfusion-rpi-setup #66: Possible fix to allow software updates in 32-bit systems

lucianomartin, 2023-03-28:
It has been noticed that when updating the software in 32-bit Raspbian versions, the system boots up in 64-bit mode and tries and fails to install the kernel headers for 64-bit modules. 
@tomblackie-xmos  found out if you set `arm_64bit=0` in the /boot/config.txt file this problem is fixed.

lucianomartin, 2023-03-30:
This solution is not working for me. I tried the following after booting a fresh SD card with the latest 32-bit Raspbian version (2023-02-21):
  1. do the Software Update at the first boot up, add `arm_64bit=0` in `/boot/config.txt` and install vocalfusion-rpi-setup for xvf3800-intdev
  2. skip the Software Update at the first boot up, add `arm_64bit=0` in `/boot/config.txt`, run `sudo apt update` and `sudo apt upgrade`,  and install vocalfusion-rpi-setup for xvf3800-intdev
  3.   skip the Software Update at the first boot up, add `arm_64bit=0` in `/boot/config.txt`, reboot the Pi, run `sudo apt update` and `sudo apt upgrade` and install vocalfusion-rpi-setup for xvf3800-intdev

In all the three cases above the v8 kernels were installed and the I2S drivers failed to compiled.

@tomblackie-xmos, am I doing something wrong?

lucianomartin, 2023-03-30:
After a discussion with @tomblackie-xmos, we found out that the correct sequence is:

1. upgrade the software, either during the first bootup or with `sudo apt update/upgrade`
2. reboot the Pi
3. update the `boot/config.txt` using the command:
`sudo sh -c "echo 'arm_64bit=0' >> /boot/config.txt"`
4. reboot the device
5. install the vocalfusion-rpi-setup

phenixxmos, 2023-03-31:
So test if this was the 32bit I used a Pi 3. I tried without an update and it failed on Pi 3. When I updated I got it to build. we should remove the box "Warning: ThefollowingstepbuildsakernelmodulefortheI2Sinterface.InsomecasestheRaspberry Pi OS upgrade process does not correctly install the required packages so XMOS currently recommends that users do not upgrade the OS for this evaluation." Since, we need to update the Pi to make it work it. I will be strictly follow the instruction to simulate the customer experience. Maybe checking if it is a Pi 4 to enable this option?

phenixxmos, 2023-04-14:
5.4.3
Got error building
Just change all ‘python’ to ‘python3’ should resolve this issue and not cause issues on other systems. This is a mac specific but won’t harm other systems

5.6.1
Scripts is not in the location specified 
Also failed on spi boot section
“python3 generate_image.py '/Users/phenix/Downloads/XM-014772-SM-4-XVF3800-INT-Device Binary/application_xvf3800_intdev-lr16-lin-i2c.xe'
Extract the binaries from the .xe file
Error detected while attempting to load file /Users/phenix/Downloads/XM-014772-SM-4-XVF3800-INT-Device: Could not open input file
Traceback (most recent call last):
  File "/Users/phenix/Downloads/XM-014771-SM-4-XVF3800 Software/sources/boot_spi_slave/scripts/generate_image.py", line 54, in run_cmd
    subprocess.check_output(cmd_to_run.split(" "))
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xobjdump', '--strip', '--split', '/Users/phenix/Downloads/XM-014772-SM-4-XVF3800-INT-Device', 'Binary/application_xvf3800_intdev-lr16-lin-i2c.xe']' returned non-zero exit status 1.
Run again with --verbose for full error”

lucianomartin, 2023-06-13:
Fixed by in v5.8.0

closed 2023-06-13 by lucianomartin