xmos/vocalfusion-rpi-setup #70: i2s build error on XVF3500 when using raspberry pi compute module 4

JihunDev, 2023-04-26:
Raspberry Pi Compute Module 4 was used and the latest version of the OS was installed.

The compute model I use is a model with 2 gigabytes of RAM and 8 gigabytes of eMMC.

If you type `./setup.sh xvf3500` command, it will output that i2s_slave_loader.c does not exist.

Tried to use the 2020 OS on the Compute model, but it won't boot.

is there any workaround?

lucianomartin, 2023-05-24:
Hi @JihunDev, thanks for reporting this issue. We have recently updated the procedure to configure the RaspberryPi. Can you please retry using the instructions in the [README.md](https://github.com/xmos/vocalfusion-rpi-setup/blob/master/README.md) at the head of master or from the [ v5.8.0 release](https://github.com/xmos/vocalfusion-rpi-setup/blob/master/README.md).

JihunDev, 2023-06-02:
Hi @lucianomartin Tried with the changed installation method and works fine on Pi4. Thank you.

closed 2023-06-02 by JihunDev