xmos/vocalfusion-rpi-setup #71: Evaluate operation Raspberry Pi OS version for compatibility with this repo

ed-xmos, 2023-05-05:


mbanth, 2023-05-05:
@ed-xmos, please provide a Story Point value for this issue.  Thanks.

nanosonde, 2024-07-30:
Hi @ed-xmos and @mbanth !

Just wanted to mentioned that I did some testing with the latest Raspberry Pi OS on a Raspberry Pi 3A+ in 64-Bit mode.

```
user@wyoming-satellite:~/vocalfusion-rpi-setup $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
user@wyoming-satellite:~/vocalfusion-rpi-setup $ uname -a
Linux wyoming-satellite 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```

The setup script worked normally.
However, there is a small glitch due to the fact that `python3-smbus` is not listed as package to be installed.
I found that out by running the crontab commands manually. 
So the DAC init was missing.

Now I can successfully run the following commands: 
`arecord -D plughw:CARD=sndrpisimplecar,DEV=0 -r 48000 -c 1 -f S32_LE -t wav -d 5 test.wav`

and then:
`aplay -Dplughw:CARD=sndrpisimplecar,DEV=0 test.wav`

However, there seems to be some issue the sound card and the HW params:

If a run `arecord -D hw:CARD=sndrpisimplecar,DEV=0 -r 48000 -c 2 -f S32_LE -t wav -d 5 test.wav`, then playing via `aplay -D VocalFusion test.wav` does not work. Pleas note that the recording was initialted on the HW pcm without any software conversions.

The output of  `aplay -D VocalFusion test.wav` looks like this:
```
ALSA lib pcm_direct.c:1336:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib pcm_dmix.c:1011:(snd_pcm_dmix_open) unable to initialize slave
aplay: main:831: audio open error: Invalid argument
```

Trying to play via this way: `aplay -D i2s_48k  test.wav`leads to:
```
Playing WAVE 'test.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: ALL
FRAME_BITS: ALL
CHANNELS: 2
RATE: [48000 48001)
PERIOD_TIME: [125000 125000]
PERIOD_SIZE: [0 0]
PERIOD_BYTES: ALL
PERIODS: ALL
BUFFER_TIME: [500000 500000]
BUFFER_SIZE: [4294967295 4294967295]
BUFFER_BYTES: ALL
TICK_TIME: [0 0]
```

I am an unsure if this is related to the I2S kernel module.
Studying the kernel docs it seems that such a simple sound card is nowadays configured solely by a device tree overlay:
https://www.kernel.org/doc/Documentation/devicetree/bindings/sound/simple-card.txt

I would be have happy if the latest RPi OS in 64-Bit could be supported by XMOS.
There are some projects for wakeword detection that require a 64-bit RPI OS.