xmos/vocalfusion-rpi-setup #72: Raspberry PI 4 , ./setup.sh was finished well but sound card wasn't added.

jason-sepion, 2023-05-18:
Raspberry Pi 4

OS : 2023-02-21-raspios-bullseye-armhf.img , From vocalfusion-rpi-setup github OS link
Kernel : Linux raspberrypi 5.15.84-v7l+
Kernel Header : linux-headers-5.15.84-v7l+

sudo ./setup.sh xvf3500 was installed well. No error
      ```
      pi@raspberrypi:~/vocalfusion-rpi-setup $ sudo ./setup.sh xvf3500
      Installing necessary packages for dev kit
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      python3-matplotlib is already the newest version (3.3.4-1).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      python3-numpy is already the newest version (1:1.19.5-1).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      libatlas-base-dev is already the newest version (3.10.3-10+rpi1).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      audacity is already the newest version (2.4.2~dfsg0-5).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      libreadline-dev is already the newest version (8.1-1).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      Reading package lists... Done
      Building dependency tree... Done
      Reading state information... Done
      libncurses-dev is already the newest version (6.2+20201114-2+deb11u1).
      The following package was automatically installed and is no longer required:
       libfuse2
      Use 'sudo apt autoremove' to remove it.
      0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
      make
      # copy the source file to avoid maintaining multiple copies inside the Makefile folders
      # We tried to use KBUILD_OUTPUT but the module was not compiling
      cp ../src/loader.c i2s_slave_loader.c
      make -C /lib/modules/5.15.84-v7l+/build M=/home/pi/vocalfusion-rpi-setup/loader/i2s_slave modules
      make[1]: Entering directory '/usr/src/linux-headers-5.15.84-v7l+'
       CC [M] /home/pi/vocalfusion-rpi-setup/loader/i2s_slave/i2s_slave_loader.o
       MODPOST /home/pi/vocalfusion-rpi-setup/loader/i2s_slave/Module.symvers
       LD [M] /home/pi/vocalfusion-rpi-setup/loader/i2s_slave/i2s_slave_loader.ko
      make[1]: Leaving directory '/usr/src/linux-headers-5.15.84-v7l+'
      rm i2s_slave_loader.c
      To enable all interfaces, this Raspberry Pi must be rebooted.
      
      ``` 

But there is no added sound card "sndrpisimplecar".

      ```
    pi@raspberrypi:~ $ aplay -l
    **** List of PLAYBACK Hardware Devices ****
    card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
     Subdevices: 1/1
     Subdevice #0: subdevice #0
    card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
     Subdevices: 1/1
     Subdevice #0: subdevice #0
    ```

I already checked the sound card "sndrpisimplecar" was added well in **Raspberry Pi 3 at card 1 .**
But in Raspberry Pi 4 it doesn't work.
There is a difference that Hdmi sound card number between Pi 3 and 4,   pi 3 have only card 0 : for hdmi.
i am not sure it is the reason.

Thanks if you can advise me how to solve this problem.

mbanth, 2023-05-23:
@lucianomartin, please provide a Story Point value for this issue.  Thanks.

lucianomartin, 2023-05-24:
Hi @jason-sepion, thanks for reporting this issue. We have recently updated the procedure to configure the RaspberryPi. Can you please retry using the instructions in the [README.md](https://github.com/xmos/vocalfusion-rpi-setup/blob/master/README.md) at the head of master or from the the[ v5.8.0 release](https://github.com/xmos/vocalfusion-rpi-setup/blob/master/README.md).

If this doesn't work, can you please let us know if the .asoundrc file is copied to the home directory?

jason-sepion, 2023-05-25:
@lucianomartin  I will follow the instructions and let you know the results right away. It may take several days. Thank you for supports.

jaritondo, 2025-03-01:
I am also running into this same issue, even when following the new v5.8.0 release. Has a solution been found for this? And will the KBUILD_OUTPUT message affect the vocal fusion setup?

mbanth, 2025-03-03:
Thanks, @jaritondo, for your question.  We will have one of our customer engineers look into the setup using v5.8.0 on a Raspberry Pi 4.